Seeed-Studio/ArduinoCore-samd #3: I2C not working on D4/D5 (SAMD21 PA8/PA9)

stanleyseow, 2020-04-16:
I encounter this issue on I2C bus and when I search issues on the main repo, the issue is still there...

SERCOM I2C not working on pins PA08, PA09 #383
https://github.com/arduino/ArduinoCore-samd/issues/383

LynnL4, 2020-04-17:
Hi
Can you provide more details? Because in our test example, most I2C devices are working properly.

stanleyseow, 2020-04-17:
I tested the following sensors : 
ADXL345 is working 

MLX90615 not working
SHT31 not working 

Can you have some sample codes too or they are just the same ??

LynnL4, 2020-04-20:
HI
  Which platform can these devices work on? And ,can you provide the URL of the library you are using?

stanleyseow, 2020-04-29:
MLX90615 https://github.com/Seeed-Studio/Digital_Infrared_Temperature_Sensor_MLX90615
SHT31 https://github.com/closedcube/ClosedCube_SHT31D_Arduino

ytsuboi, 2020-05-21:
@LynnL4 bump

closed 2021-05-08 by Pillar1989