Seeed-Studio/ArduinoCore-samd #4: error: macro "min" passed 3 arguments at stl_algobase.h

matsujirushi, 2020-04-29:
I want to use STL.
This code got error.

**Board:** Seeeduino XIAO

**Code:**
```
#include <vector>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

**Error message:**
```
In file included from c:\users\takashi\documents\arduinodata\packages\seeeduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\vector:60:0,

                 from C:\Users\takashi\AppData\Local\Temp\arduino_modified_sketch_889764\sketch_apr29a.ino:1:

c:\users\takashi\documents\arduinodata\packages\seeeduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\stl_algobase.h:239:56: error: macro "min" passed 3 arguments, but takes just 2

     min(const _Tp& __a, const _Tp& __b, _Compare __comp)

                                                        ^

c:\users\takashi\documents\arduinodata\packages\seeeduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\bits\stl_algobase.h:260:56: error: macro "max" passed 3 arguments, but takes just 2

     max(const _Tp& __a, const _Tp& __b, _Compare __comp)

                                                        ^

exit status 1
```

LynnL4, 2020-05-03:
Try to undefine the minimum before including STL.
```C++
#undefine min
#undefine max
#include <stl.h>
```

LynnL4, 2020-05-07:
This problem has been fixed.

closed 2020-05-07 by LynnL4

matsujirushi, 2020-05-07:
https://github.com/Seeed-Studio/ArduinoCore-samd/commit/b26011576f5395bcac3d56b0d48ef80b666a3f4d