Seeed-Studio/ArduinoCore-samd #12: warning: "LITTLE_ENDIAN" redefined

mnakai3, 2020-05-13:
When I built Seeeduino XIAO with v1.7.5, the following warning is displayed.

```
"C:\\...\\Arduino15\\packages\\Seeeduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD "-D__SKETCH_NAME__=\"\"\"Blink.ino\"\"\"" -DF_CPU=48000000L -DARDUINO=10812 -DARDUINO_SEEED_XIAO_M0 -DARDUINO_ARCH_SAMD -DARDUINO_SAMD_ZERO -D__SAMD21__ -D__SAMD21G18A__ -DARM_MATH_CM0PLUS -DSEEED_XIAO_M0 -DUSB_VID=0x2886 -DUSB_PID=0x802F -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Seeed\"" "-DUSB_PRODUCT=\"Seeed XIAO M0\"" -g "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -g -DARDUINO_SAMD_ZERO -D__SAMD21__ -D__SAMD21G18A__ -DARM_MATH_CM0PLUS -DSEEED_XIAO_M0 -DUSB_VID=0x2886 -DUSB_PID=0x802F -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Seeed\"" "-DUSB_PRODUCT=\"Seeed XIAO M0\"" -g "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino" "-IC:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\variants\\XIAO_m0" "C:\\...\\Arduino15\\packages\\Seeeduino\\hardware\\samd\\1.7.5\\cores\\arduino\\main.cpp" -o "C:\\...\\Temp\\arduino_build_625793\\core\\main.cpp.o"
In file included from C:\...\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,
                 from C:\...\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd.h:105,
                 from C:\...\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/sam.h:540,
                 from C:\...\Arduino15\packages\Seeeduino\hardware\samd\1.7.5\cores\arduino\SERCOM.h:22,
                 from C:\...\Arduino15\packages\Seeeduino\hardware\samd\1.7.5\cores\arduino\Uart.h:22,
                 from C:\...\Arduino15\packages\Seeeduino\hardware\samd\1.7.5\cores\arduino\Uart.cpp:19:
C:\...\Arduino15\packages\Seeeduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:226:0: warning: "LITTLE_ENDIAN" redefined

 #define LITTLE_ENDIAN          1
```

Environment
=========
- Arduino IDE: 1.8.12
- sketch: 01.Basics / Blink

matsujirushi, 2020-05-15:
I think it's the wrong definition 'LITTLE_ENDIAN' in samd21g18a.h.
@LynnL4  Could you tell me where do you control the CMSIS sources? (github?)

[image]

LynnL4, 2020-05-15:
This problem occurs in the CMSIS-Atmel provided by ATMEL. The use of this file is closely related to the hardware. So we need to modify it carefully

Pillar1989, 2020-05-20:
https://github.com/Seeed-Studio/Seeed_Arduino_CMSISAtmel @matsujirushi  @mnakai3  We have updated and the next version USES CMSIS above GitHub. @LynnL4  Please help check it.

matsujirushi, 2020-05-20:
https://github.com/Seeed-Studio/Seeed_Arduino_CMSISAtmel/commit/ace7dc949c0737c63eaa903103b8ce234c54bd6a

mnakai3, 2020-05-21:
Hi @Pillar1989 ,
I confirmed commit.
The header referenced when building XIAO is "samd21g18a.h".
Could you also modify this file?

ytsuboi, 2020-05-21:
@Pillar1989 Please proceed testing before push.

Pillar1989, 2020-05-21:
You're right.

Pillar1989, 2020-05-21:
https://github.com/Seeed-Studio/Seeed_Arduino_CMSISAtmel/commit/dcc4a2770d33c89778b033c419e18588ba87bae9 I tested this commit yesterday, and you can use a higher version of the compiler.I tested the gcc-9. @ytsuboi   @mnakai3

ytsuboi, 2020-05-21:
@Pillar1989 The header file you modified was sam**r**21g18a.h, not XIAO target samd21g18a.h. So that we doubt your testing.

ytsuboi, 2020-05-21:
Let's close this issue after the confirmation. Where's gcc9 package you used?

Pillar1989, 2020-05-21:
On our internal test system, You can also download it here. 
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

closed 2020-05-29 by LynnL4