Seeed-Studio/ArduinoCore-samd #13: Serial.begin() return before begins communication with USB host.

ytsuboi, 2020-05-20:
Work around is: https://gist.github.com/ciniml/bd2bb4dea22a55c067ca6790ad0c3eae#file-grove_left_uart-ino-L7

Nice to have while loop in Serial.begin().

Pillar1989, 2020-05-21:
@ytsuboi   USB Serial `while(!Serial);` is currently available. Other SERCOM's don't make sense without flow control.

ytsuboi, 2020-05-21:
@Pillar1989 I mean
```
void setup() {
  Serial.begin(115200);
  delay(2000);
  Serial.println("setup");
}

void loop() {
  Serial.println("loop");
  delay(1000);
}
```
Users can't get "setup" and "loop" on serial monitor. ( Somehow, "loop" can get when they use other serial terminal software. ) To solve this, people need to write like
```
void setup() {
  Serial.begin(115200);
  while(!Serial);
  Serial.println("setup");
}

void loop() {
  Serial.println("loop");
  delay(1000);
}
```

This is simply because Serial.begin() returned before CDC begins communication with host. Serial.begin() should returns when it get begin.

LynnL4, 2020-05-29:
It is recommended that the user manually control by while(!Serial);,  because in some application scenarios, the user does not need to wait for the serial monitor opened before executing the subsequent program.

closed 2020-05-29 by LynnL4