Seeed-Studio/ArduinoCore-samd #24: Seeduino LoRaWAN SPI - SD Card Library

benbodan, 2020-07-24:
It seems that the SPI it's not working properly with the SD Card Library. For Example I  run the CardInfo.ino example on another Arduino Clone and it works fine , when I run the same code on Seeeduino LoRaWAN the card can't be initialized same thing happens with other SPI Peripherals for example ADC (mcp3008). Almost two years ago it used to work on Seeeduino LoRaWAN too.

jiacheng-lu, 2020-08-03:
The hardware structure of lorawan is different from other Zero series, so we have to change the pin configuration code in the software, so that lorawan can be used later.

LynnL4, 2020-08-05:
Hi,
[dev-1.7.9](https://github.com/Seeed-Studio/ArduinoCore-samd/tree/dev-1.7.9) May help you. 
Please replace the original ArduinoCore with the following command:
`git clone -b dev-1.7.9 https://github.com/Seeed-Studio/ArduinoCore-samd 1.7.9 --recursive`

benbodan, 2020-08-05:
Hello , 
I replaced the ArduinoCore-samd with the version 1.7.9 but it  doesn't solved the problem either.

Using library SPI at version 1.0 in folder: C:\Users\benbo\AppData\Local\Arduino15\packages\Seeeduino\hardware\samd\1.7.9\libraries\SPI

LynnL4, 2020-08-06:
Hi,
This is very strange. In our test, SPI is working. Can it be the problem of CS? Can you provide more details? We want to reproduce this problem

benbodan, 2020-08-07:
Using a Logic Analyzer I can see that the SPI pins are working , I will check the v 1.7.9 with a different SD Card tomorrow , right now for some reason the SD Card stopped  working with my other Arduino clone  too . I am sure it's not the CS because the Card Info with the same shield worked on my other Arduino uno clone one week before.

benbodan, 2020-08-10:
I just tested another SD Card Shield with an  Arduino Uno clone is initializing the SD Card using cardInfo example with chipSelect = 10, but with Seeduino LoRa WAN it won't initialize the Card . 

The new shield is using the pins below :
13 SCK
12 MISO
11 MOSI
10 SS

LynnL4, 2020-08-14:
Hi，
In fact, the SPI of the zero series is not in pin 13 12 11 & 10. So you When I run SD card library directly, it works.
```
Initializing SD card...Wiring is correct and a card is present.

Card type:         SDHC
Clusters:          485936
Blocks x Cluster:  64
Total Blocks:      31099904

Volume type is:    FAT32
Volume size (Kb):  15549952
Volume size (Mb):  15185
Volume size (Gb):  14.83

Files found on the card (name, date and size in bytes): 

```

closed 2020-08-14 by LynnL4