Seeed-Studio/ArduinoCore-samd #40: Cannot set interrupt priority correctly for FreeRTOS environment.

ciniml, 2020-12-18:
`CMSIS-Atmel` in the current toolchain used by ArduinoCore-samd has a bug which the "atsamd51p19a.h" contains wrong definition of  `__NVIC_PRIO_BITS`.

This causes that applications can not use interrupt priority other than 3, which limits the interrupt usage in applications.

The description of  the mechanism is written in [my article](https://zenn.dev/ciniml/articles/wioterminal_freertos) (Sorry, it's only in Japanese. )

The code to reproduce the problem is [here](https://github.com/ciniml/wioterminal_freertos_bug)

In order to fix this issue, please change the definition of `__NVIC_PRIO_BITS` from 2 to 3 in "atsamd51p19a.h" in `CMSIS-Atmel`

(Updating `CMSIS-Atmel` also solves the issue, but the structure of `CMSIS-Atmel` is drastically changed from current version and it's very difficult to update `CMSIS-Atmel`. So I think just changing `__NVIC_PRIO_BITS` is the easiest way to solve the issue.)

LynnL4, 2020-12-21:
Thank you! We will review ASAP.

LynnL4, 2020-12-21:
https://github.com/Seeed-Studio/Seeed_Arduino_CMSISAtmel/commit/87a1dce4e0603e6d522deb5302df8ac740041dfb. This problem has been fixed and will be updated in the next release. Thanks!