Seeed-Studio/ArduinoCore-samd #55: On seeeduino XIAO analogWrite() to A4 copies to A10 and vice-versa. Same for A5 and A8.

haiershu, 2021-12-02:
The attached video shows PROGRAM #1 simp[le loop turning on / off each LED using digitalWrite() and PROGRAM #2 using analogWrite().

analogWrite() causes crosstalk between A5 and A8 and also between A4 and A10. Please watch the attached video.

Here’s the schematic for the test setup:

Here’s the code used to demonstrate the XIAO problem.

/*

PROGRAM #1

Demonstrate XIAO digitalWrite()

does NOT crosstalk

*/

void setup() {

for(int i = 1; i <= 10; i++) {

pinMode(i, OUTPUT);

digitalWrite(i, HIGH);

}

}

void loop() {

for(int i = 1; i <= 10; i++) {

digitalWrite(i, LOW);

delay(1000);

digitalWrite(i, HIGH);

}

delay(1000);

}

_______________________________________

/*

PROGRAM #2

Demonstrate XIAO analogWrite()

DOES crosstalk

*/

void setup() {

for(int i = 1; i <= 10; i++) {

pinMode(i, OUTPUT);

analogWrite(i, 255);

}

}

void loop() {

for(int i = 1; i <= 10; i++) {

analogWrite(i, 0);
delay(1000);

analogWrite(i, 255);

}

delay(1000);

}

haiershu, 2021-12-02:
Test environment: windows 10
Arduino IDE version: 1.8.3

haiershu, 2021-12-02:
https://seeed.uservoice.com/assets/233417448/XIAO_analogWrite_Crosstalk_Problem.avi

LynnL4, 2021-12-21:
Hi, due to the resource limitation of the chip, some pins are multiplexed to achieve the function, so it will lead to the current situation. My suggestion is that the function pins (I2C SPI...) are not used as analog pins

closed 2021-12-21 by LynnL4