Seeed-Studio/ArduinoCore-samd #68: Unable to read SPI data from slave

silentjet, 2023-07-15:
Hi,
I'm trying to read the software version from peripheral device as a sanity check upon software startup and faced with very strange issue. I wrote a bare-bone code to check that, and it still has the same issue.

I'm sending a command 0x70 and I expect two bytes of response. Using a logic analyzer I can clearly see the response arrives and values are meaningful. But at the same time what is returned by `SPI.transfer()` it is zero instead of visible on analyzer values. I look through the library code and found out something like that
https://github.com/Seeed-Studio/ArduinoCore-samd/blob/v1.8.4/cores/arduino/SERCOM.cpp#L362
```cpp
uint8_t SERCOM::transferDataSPI(uint8_t data)
{
  sercom->SPI.DATA.bit.DATA = data; // Writing data into Data register

  while(sercom->SPI.INTFLAG.bit.RXC == 0); // Waiting Complete Reception

  return sercom->SPI.DATA.bit.DATA;  // Reading data
}
```
I do not have too much knowledge about that, but I assume it shall work properly. But in reality it doesn't.

[image]

Lesords, 2024-10-16:
Hello,

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

silentjet, 2024-10-17:
Hello,
Sorry, can't say, cause I switched to another board (riscv based xaio).

closed 2025-01-30 by github-actions[bot]