Seeed-Studio/ArduinoCore-samd #73: Broken defines for SerialUSB, PIN_LED_RXL and PIN_LED_TXL

nichtgedacht, 2025-05-06:
My Project with an XIAO_m0 is sticking at 1.8.3 because of these missing defines.

Suggested simple Patch (which surely can't break anything):

`
diff --git a/variants/XIAO_m0/variant.h b/variants/XIAO_m0/variant.h
index ffc9cf6..17ea50e 100644
--- a/variants/XIAO_m0/variant.h
+++ b/variants/XIAO_m0/variant.h
@@ -74,6 +74,9 @@
 #if defined(TXRXLED_ENABLE)
 #define PIN_LED_RXL          PIN_LED2
 #define PIN_LED_TXL          PIN_LED3
+#else
+#define PIN_LED_RXL          (12U)
+#define PIN_LED_TXL          (11U)
 #endif
 
 /*
@@ -219,3 +222,5 @@ extern Uart Serial1;
 #define SERIAL_PORT_HARDWARE_OPEN   Serial1
 
 #endif /* NO_USART_INTERFACE */
+
+#define Serial SerialUSB
`
Regards

Lesords, 2025-05-19:
Hello,

I'm curious what problem you encountered?

The changes you made will cause the RX/TX LED options to be invalid.

nichtgedacht, 2025-05-19:
Hi,
 
this Project does not compile because of missing defines of  PIN_LED_TXL,  PIN_LED_RXL and SerialUSB if not keeping the old version of the framework.:
https://github.com/nichtgedacht/JetiMiniVario2/
So I’m stuck at those version which was used at the begin of my project: #991c4b032270cc7e53fdd70e31e75eeca2bdedc8

 

 
 
Gesendet: Montag, 19. Mai 2025 um 08:51

Lesords, 2025-05-20:
I see, you are using platformio.

I made some modifications based on your repository, and it works now.

Add the following to the end of the `platformio.ini` file
```ini
build_flags = 
    -DTXRXLED_ENABLE
```

Add the following to the `config.h` file
```cpp
#define SerialUSB Serial
```

Lesords, 2025-05-20:
And, I would not add code like this, as it would render the RX/TX LED options useless.

```cpp
#if defined(TXRXLED_ENABLE)
#define PIN_LED_RXL PIN_LED2
#define PIN_LED_TXL PIN_LED3
#else
#define PIN_LED_RXL (12U)
#define PIN_LED_TXL (11U)
#endif
```

Lesords, 2025-05-20:
> I see, you are using platformio.
> 
> I made some modifications based on your repository, and it works now.
> 
> Add the following to the end of the `platformio.ini` file
> 
> build_flags = 
>     -DTXRXLED_ENABLE
> Add the following to the `config.h` file
> 
> #define SerialUSB Serial

Of course, these are all changes made after deleting `#991c4b032270cc7e53fdd70e31e75eeca2bdedc8`

nichtgedacht, 2025-05-20:
Thank you for the workaround. However, I am forced to make changes to my project due to the changes to the framework. This should not be the case. I will stick with the frozen version so that nothing unexpected happens.